Legion2/CorsairLightingProtocol #295: Compiler Errors on example project, did something break?

HasseM, 2022-08-11:
**Describe the problem**
New PC. setting up my dev environment. Trying to run this project on a RaspberryPi Pico. It was all working on my old PC. RPi pico is still connected and recognized by iCue. Now I want to get it all working on my new PC but I end up in a large amount of errors. I have been trying to fix it for a couple of hours now and I seem to be missing something..... or did it break with new versions of stuff?

**Screenshots**
`In file included from q:\arduino\libraries\FastLED\src/FastLED.h:50,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:34,
                 from C:\Users\username\AppData\Local\Temp\.arduinoIDE-unsaved202279-11936-1w9aq2j.asp6\LightingNodePRO\LightingNodePRO.ino:16:
q:\arduino\libraries\FastLED\src/fastpin.h:120:37: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  120 |  inline void hi(register port_ptr_t port) __attribute__ ((always_inline)) { *port |= mPinMask; }
      |                                     ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:121:37: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  121 |  inline void lo(register port_ptr_t port) __attribute__ ((always_inline)) { *port &= ~mPinMask; }
      |                                     ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:122:34: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  122 |  inline void set(register port_t val) __attribute__ ((always_inline)) { *mPort = val; }
      |                                  ^~~
q:\arduino\libraries\FastLED\src/fastpin.h:124:42: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  124 |  inline void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                          ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:124:64: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  124 |  inline void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                                                ^~~
q:\arduino\libraries\FastLED\src/fastpin.h:228:44: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  228 |  inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { }
      |                                            ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:229:44: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  229 |  inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { }
      |                                            ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:230:41: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  230 |  inline static void set(register port_t val) __attribute__ ((always_inline)) { }
      |                                         ^~~
q:\arduino\libraries\FastLED\src/fastpin.h:232:49: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  232 |  inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { }
      |                                                 ^~~~
q:\arduino\libraries\FastLED\src/fastpin.h:232:71: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  232 |  inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { }
      |                                                                       ^~~
In file included from q:\arduino\libraries\FastLED\src/FastLED.h:51,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:34,
                 from C:\Users\username\AppData\Local\Temp\.arduinoIDE-unsaved202279-11936-1w9aq2j.asp6\LightingNodePRO\LightingNodePRO.ino:16:
q:\arduino\libraries\FastLED\src/fastspi_types.h:22:82: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   22 |     static __attribute__((always_inline)) inline uint8_t adjust(register uint8_t data) { return data; }
      |                                                                                  ^~~~
q:\arduino\libraries\FastLED\src/fastspi_types.h:23:82: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   23 |     static __attribute__((always_inline)) inline uint8_t adjust(register uint8_t data, register uint8_t scale) { return scale8(data, scale); }
      |                                                                                  ^~~~
q:\arduino\libraries\FastLED\src/fastspi_types.h:23:105: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   23 |     static __attribute__((always_inline)) inline uint8_t adjust(register uint8_t data, register uint8_t scale) { return scale8(data, scale); }
      |                                                                                                         ^~~~~
In file included from q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastled_arm_rp2040.h:5,
                 from q:\arduino\libraries\FastLED\src/platforms.h:35,
                 from q:\arduino\libraries\FastLED\src/FastLED.h:54,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18,
                 from q:\arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:34,
                 from C:\Users\username\AppData\Local\Temp\.arduinoIDE-unsaved202279-11936-1w9aq2j.asp6\LightingNodePRO\LightingNodePRO.ino:16:
q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:28:42: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   28 |   inline static void set(register port_t val) __attribute__ ((always_inline)) { sio_hw->gpio_out = val; }
      |                                          ^~~
q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:34:45: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   34 |   inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { hi(); }
      |                                             ^~~~
q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:35:45: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   35 |   inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { lo(); }
      |                                             ^~~~
q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:36:50: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   36 |   inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port = val; }
      |                                                  ^~~~
q:\arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:36:72: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
[... 121 lines cut ...]
   21 |   register uint32_t bn=0;
      |                     ^~
q:\arduino\libraries\fastled\src\platforms\arm\common\m0clockless.h:22:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   22 |   register uint32_t b=0;
      |                     ^
q:\arduino\libraries\fastled\src\platforms\arm\common\m0clockless.h:23:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   23 |   register uint32_t bitmask = _bitmask;
      |                     ^~~~~~~
q:\arduino\libraries\fastled\src\platforms\arm\common\m0clockless.h:26:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   26 |   register const uint8_t *leds = _leds;
      |                           ^~~~

exit status 1

Compilation error: 'CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?`

**System (please complete the following information):**
 - OS: [Win11]
 - Board: [RapsberryPi Pico]
 - Device: [Lighting Node PRO]
 - Version of IDE: [1.8.19 but also tried latest 2.0 RC9]

**Code changes**
Did you changed the code of the library or the sketch?
No.... just the example project....

HasseM, 2022-08-15:
An update from my side:

I decided to brake down my problems in smaller steps. 

1) does the board/setup work
So I first set to get a blinky example running on my RPico. This failed because my new pc environment had never seen a serial port from a working device yet. See: https://arduino-pico.readthedocs.io/en/latest/install.html#unable-to-upload-first-sketch  So I just couldnt select the port. Probably nothing today with above problem because that already happens at compile. But anyhow this could be easily solved and now I am sure my Pico Arduino environment works. So not board issues.

Still the same issue as above. Any got advice?

HasseM, 2022-08-15:
Ok, found a problem in the code:

In CorsairLightingFirmwareStorageEEPROM.h there is the following line (Ln18):
#if defined(ARDUINO_ARCH_AVR)

When using a RaspberryPi Pico, this is ARDUINO_ARCH_RP2040 and not ARDUINO_ARCH_AVR. Or is it me doing something wrong?

Fixing this files jumps to a similar problem on the next line. I can easily now go through the code and fix all the #if deffined statements (using the OR statement). But waiting for advice here!

Legion2, 2022-08-15:
Hi, `CorsairLightingFirmwareStorageEEPROM` is not available for pi pico, try to use the static storage instead

HasseM, 2022-08-16:
> Hi, `CorsairLightingFirmwareStorageEEPROM` is not available for pi pico, try to use the static storage instead

So you mean the LightningNodePro example is not working for the Pico? Note that I am following exact step as in the readme..... [here](https://github.com/Legion2/CorsairLightingProtocol#create-a-lighting-node-pro-for-a-raspberry-pi-pico-with-tinyusb)

If I replace the Eeprom referemces to StorageStatic I get the following error:
Compilation error: no matching function for call to 'CorsairLightingFirmwareStorageStatic::CorsairLightingFirmwareStorageStatic()

Legion2, 2022-08-16:
In the readme the Pi Pico step by step guide uses the `TinyUSB` example instead of the `LightningNodePro`.

HasseM, 2022-08-16:
arggggg...... it's always that small reading mistake that you make that keeps you busy for hours.......

thanks! It's all clear now!

closed 2022-08-18 by HasseM