Legion2/CorsairLightingProtocol #296: A couple of small things

HasseM, 2022-08-18:
Using a RPi Pico to make a lightning node pro compatible led show,

I ran into a couple of small things that I cannot find the answers for. Hopefully someone here can give a little push in the right direction.

- How to detect hardware mode? Currently the running software effect just stops updating (e.g. when I lock my PC). So how to detect and set my own hardware led states? Anything else then just stopping the running show would already help me here!

- How to delay led updates? For example when I make a custom effect in Arduino, not based on iCue data at all, I want to control the timing/speed of my effect. Whats the best way to do that?

- Additional to the above: I probably dont want to disable the update hook because I use an unused led pixel to detect which 'mode' my project must visualize. I guess there is no other way to pass settings from iCue to my project. Right?

- Is there a direct way to receive a temperature value from iCue in the Arduino? Or should I set up a ledstrip temperature gradient and derive it from the color value?

- How to map to intermediate colors (probably more a fastled question though)? E.g. If iCue sends data for a 15 led ledstrip that I want to map to my project, the rounding down to discrete values causes strange side effects. As such I want to calculate inbetween values. E.g. when mapping to 9 leds, led #8 should have the value of iCue Led #13,33333  (8/9 * 15). Anyone knows a nice example doing that?

HasseM, 2022-08-18:
For my first issue (detecting hardware mode) I would like to propose a code change:

FastLEDController.h Ln 99 (just below the getLEDCount declaration):
```
	/**
	 * Get the channel mode.
	 *
	 * @param channel the index of the channel
	 * @return the channel mode: 0=disabled,1=hardware,2=software
	 */
	uint8_t getChannelMode(uint8_t channel);
```

And in FastLEDController.cpp Ln61 add the following function:
```
uint8_t FastLEDController::getChannelMode(uint8_t channel){
	if (channel >= CHANNEL_NUM) {
		return 0;
	}
	LEDChannel& chan = channels[channel];
	if (chan.mode == ChannelMode::Disabled) return 0;
	if (chan.mode == ChannelMode::HardwarePlayback) return 1;
	if (chan.mode == ChannelMode::SoftwarePlayback) return 2;
	return 0;
}
```

Legion2, 2022-08-19:
1. Can you explain your use case better. If you want to turn off animations if PC is locked (Hardware mode) you can configure an static black color effect in iCUE for Hardware mode
2. You can not receive temperature from iCUE, only animations in iCUE have access to that, but you can create an animation based on temperature and then read the LED RGB value to get the temperature.
3. The CLP library includes multiple scaling functions to map between different number of leds

HasseM, 2022-08-19:
1. For hardware mode, the number of animations is rather limited. So if you want to make a custom animation while in hardware mode then you need to be able to detect hardware mode. For turning off a static color black can be used (if the bug is fixed, now it doesnt work with this code base). But I would like to make some kind of heart-beat effect.

2. clear, thanks for explaining. Now I dont have to search for something that is not there.

3. Do you happen to know of some examples. It's rather hard to find my way round the different functions and possibilties.

Legion2, 2022-08-19:
1. to support hardware mode customization, https://github.com/Legion2/CorsairLightingProtocol/blob/0f451bb78589c017339645555787f0328e4d981c/src/FastLEDController.cpp#L376-L429 should be extracted into a method, which can be replaced/overwritten with a custom implementation
3. see https://github.com/Legion2/CorsairLightingProtocol#repeat-or-scale-led-channels and https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel

HasseM, 2022-08-19:
1. Yes, but I do not like changing libraries..... on each update your changes are gone.....

Legion2, 2022-08-22:
1. yes that should be something the library should support out of the box, but currently I have no time to do these changes

closed 2022-09-14 by github-actions[bot]