Legion2/CorsairLightingProtocol #297: Hardwaremode render functions are not all working

HasseM, 2022-08-18:
**Describe the bug**
Most hardwaremode render functions are not working.

**To Reproduce**
Set hardwaremode (e.g. lock your screen in windows) and see that staticcolor, color pulse, color shift are not working. Rainbow wave, Marquee do work.

I have been testing and debugging the code.
if I change renderStatic (FastLEDController.cpp Ln 202) from:
```
bool FastLEDController::renderStatic(ChannelData& channelData, LEDGroup& group, int groupLedCount) {
	fill_solid(&channelData.leds[group.ledIndex], groupLedCount, group.color1);
	return true;
}
```
to:
```
bool FastLEDController::renderStatic(ChannelData& channelData, LEDGroup& group, int groupLedCount) {
int duration = applySpeed(700, group.speed);
int count = animation_step_count(duration, 3);	
if (count > 0) {
	for (int i = 0; i < groupLedCount; i++) {	
		channelData.leds[group.ledIndex + i] = group.color1;
	}
	//fill_solid(&channelData.leds[group.ledIndex], groupLedCount, group.color1);
	return true;
	}
return false;	
}
```
Then it works. For some reasons, the animation stuff is needed for it to work. It's likely also the case for the other render functions.

Legion2, 2022-08-19:
Which arduino device are you using? Can you create a PR to fix the issue?

HasseM, 2022-08-19:
> Which arduino device are you using? Can you create a PR to fix the issue?

RaspberryPi Pico

Can you tell me how to create a 'PR to fix the issue' ? I would be happy to help here!

Legion2, 2022-08-19:
You need to fork this repository, do a git clone on the forked repository, fix the code, create a commit and the push the commit to your fork, from there you can create a Pull Request, which we can then discuss here on github.

HasseM, 2022-08-19:
Yes, can do but I am ot an expoert on this code (yet). So if you feel that my proposed change is good, I will submit it. However I am just not sure if there is'nt a better solution then forcing a static situation to perform a fake animation. I just dont have the overview yet. I am in this code just a couple of days....

closed 2024-01-30 by Legion2