Legion2/CorsairLightingProtocol #298: Add two LED strips on the same channel

legouju, 2022-09-01:
**Describe the problem**
Hello, I am trying to add two 30 LED WS2812B strips onto the same channel (ledsChannel2)

**Screenshots**
[image]

**Code changes**
The code above complies and is able to be uploaded to my Arduino, pins 9 and 11 work, but no data comes out of pin 10.
If I use the LightingnodPro example and Define data pins as 9 and 10, it works fine.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo (https://wiki.dfrobot.com/Beetle_SKU_DFR0282)
 - Device: Lighting Node PRO
 - Version of IDE: Arduino IDE 1.8.20
 - Version of CLP: 0.15.0
 - Version of CLP Boards: 0.3.1

Legion2, 2022-09-01:
Not all pins on the Arduino Leonardo can be used with FastLED, check the documentation of FastLED and your Arduino Leonardo board to verify you use pins which are supported.
Your sketch itself is similar to https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino, which also splits a channel into multiple strips and use these strips for led fans.

legouju, 2022-09-02:
Thanks for the reference, was able to separate them into multiple channels.
Also was wondering if it would be possible to reverse the direction on a certain pin.
CLP::reverse will reverse the whole channel (ie ledsChannel1), but I only want a certain data pin to be reversed, (ie DATA_PIN_STRIP_2).

Thanks again!

`#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_STRIP 30

#define DATA_PIN_STRIP_1 9
#define DATA_PIN_STRIP_2 10

#define DATA_PIN_STRIP_3 11

#define NUMBER_OF_LEDS_PER_FAN 4
#define DATA_PIN_FAN_1 A0
#define DATA_PIN_FAN_2 A1

#define DATA_PIN_FAN_3 A2

CRGB ledsChannel1[96];
CRGB ledsChannel2[96];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_STRIP * 0, NUMBER_OF_LEDS_PER_STRIP);
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_STRIP * 1, NUMBER_OF_LEDS_PER_STRIP);
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_STRIP * 2, 36);

  // normal strip on channel 2
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, 88);
  
  //FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 4);
  
  
  ledController.addLEDs(0, ledsChannel1, 96);
  ledController.addLEDs(1, ledsChannel2, 96);

}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}`

Legion2, 2022-09-03:
If the built-in function do not fit your needs, you can write your own LED data manipulation functions, which you can customize to your needs. If you need hints how this works, have a look at the built-in functions and how they work.

legouju, 2022-09-04:
Alright thanks, right now just using open RGB to change the position as needed.

Again, thanks for the advice and help!

closed 2022-09-04 by legouju