Legion2/CorsairLightingProtocol #300: 'CorsairLightingProtocolHID' does not name a type  CorsairLightingProtocolHID cHID(&cLP);

sadfsdgsg, 2022-09-16:
Hi there,

while compiling various example sketches (Lighting Node Pro and TinyUSB) i get the error message
LightingNodePRO.ino:31:1: error: 'CorsairLightingProtocolHID' does not name a type
 CorsairLightingProtocolHID cHID(&cLP);

tried reinstalling library, different sketches and so on. Anyone has a clue?

Legion2, 2022-09-16:
Did you followed the guide and installed and selected the correct CLP boards before compiling?

sadfsdgsg, 2022-09-16:
Thanks for the quick response!!
Yes, i actually tried the Uno guide and die guide for Raspi pico.
Got the same issue during compiling...
Nevertheless, i would like to focus on the issue on the raspi pico, since there are less steps for getting it to run.
I used the Rasperry Pi Pico/RP2040 Boardmanager form Earle F. Philhower Version 2.5.2
FastLED by Daniel Garcia Version 3.5.0
Corsair Lighting Protocol Version 0.15.0
Arduino IDE 2.0.0
Raspi Pico from 2020(guess V 1.2)

xyberred, 2022-09-16:
You also need to install the Adafruit TinyUSB library. After that make sure that Raspberry Pi Pico board is selected under Tools. Then under the Tools -> USB Stack, select Adafruit TinyUSB instead of the Pico SDK.

sadfsdgsg, 2022-09-18:
Okay, thanks for the answer again!
Unfortunately, i don't have enough time at the moment, to fiddle with it...
Anyway, with my freshly ordered Pro Micro everything runs smoothly, so the problem must be on my side.
Thank you for your help!!!

closed 2022-09-18 by Legion2

sadfsdgsg, 2022-09-18:
Okay... even though i do not have the time, i tried something else...
uploading as Raspberry Pi Pico does not work somehow... As "Generic RP2040" everything runs smoothly, and it gets detected by OpenRGB with both channels.
Selected USB Stack "Adafruit TinyUSB" with the above mentioned soft- and hardware...
I did not try with the leds connected, but i guess the important part is done. Many thanks for your help @xyberred and @Legion2 !!
I love the project!