Legion2/CorsairLightingProtocol #306: USB device descriptor failed after reboots

legouju, 2022-10-14:
**Describe the problem**
Sometimes when I restart my system, the Arduino fails to connect as seen below. I have to physically unplug the Arduino from my system and plug it back in for it to work again. Was wondering what would be causing this and if there's a fix for it

Looked at a similar issue where windows also threw the same error I'm getting, but unfortunately, their issue was just a loose connection since my connection is just micro b and not soldered. 
https://github.com/Legion2/CorsairLightingProtocol/issues/248

**Screenshots**
[image]

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo (Arduino beetle)
 - Device: Lighting Node PRO
 - Version of IDE: Arduino IDE 1.8.2
 - Version of CLP: 0.15.0
 - Version of CLP Boards: 0.3.1

**Code changes**
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_STRIP 30

#define DATA_PIN_STRIP_1 A0
#define DATA_PIN_STRIP_2 A1
#define DATA_PIN_STRIP_3 A2

#define NUMBER_OF_LEDS_PER_FAN 4
#define DATA_PIN_FAN_1 9
#define DATA_PIN_FAN_2 10
#define DATA_PIN_FAN_3 11
#define DATA_PIN_FAN_4 3

CRGB ledsChannel1[96];
CRGB ledsChannel2[96];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_STRIP * 0, NUMBER_OF_LEDS_PER_STRIP);
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_STRIP * 1, NUMBER_OF_LEDS_PER_STRIP);
  //FastLED.addLeds<WS2812B, DATA_PIN_STRIP_3, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_STRIP * 2, 36);

  FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel2, 0, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel2, 4 + NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel2, 4 + NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel2, 4 + NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN * 12);
  

  ledController.addLEDs(0, ledsChannel1, 96);
  ledController.addLEDs(1, ledsChannel2, 96);

}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}

Legion2, 2022-10-29:
Sorry for the late response. How often does it happen, more or less than 50% of the times? Does it also happen if you use the unmodified sample code?

legouju, 2022-10-30:
its alr

It happens pretty infrequently so it makes it hard to reproduce.

I haven't tried using the unmodified sample code to see if it happens with that, but I now have it plugged into a USB hub before going into the motherboard USB header so I will let you know if that helps.

closed 2022-11-22 by github-actions[bot]