Legion2/CorsairLightingProtocol #308: Can't scale leds

tripsterstylus, 2022-10-30:
**Describe the problem**
I have an 72 leds strip and i want to scale it to 85 to use 1.4m strip in corsair but after modifing the code , the board isn't show in Icue
Here is the code : 
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 3

CRGB ledsChannel1[72];

// Most ARM devices do not contain an EEPROM; we will use static storage for the Device ID
DeviceID deviceID = {0x9A, 0xDA, 0xA7, 0x8E};
CorsairLightingFirmwareStorageStatic firmwareStorage(deviceID);
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDController ledController(nullptr);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolTinyUSBHID cHID(&cLP);

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 72);
	ledController.addLEDs(0, ledsChannel1, 85);
	ledController.onUpdateHook(0, []() {
		CLP::scale(&ledController, 0, 72);
	});
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. PI PICO]
 - Device: [e.g. Lighting Node PRO]
 - Version of IDE: [e.g. Arduino IDE 1.8.13]
 - Version of CLP: [e.g. 0.15.0]
 - Version of CLP Boards: [e.g. 0.2.0]
 - Sketch: [RepeatAndScale]

Legion2, 2022-10-30:
in `CRGB ledsChannel1[72];` the 72 must be 85, then it should work.

tripsterstylus, 2022-10-31:
i tried but no results. Its like first time

closed 2022-11-24 by github-actions[bot]