Legion2/CorsairLightingProtocol #309: LT100 LED's random colour flicker during ambient mode

GodlyOverlord, 2022-11-01:
**Describe the problem**
I'm unsure if this is a bug or something else but my LT100's constantly exhibit an issue where the LED's will randomly flicker either between off and on or other random colours.

I am using Arduino Pro Micros and have 2 flashed as "lighting node pro"'s (PC and underdesk lighting) and 1 flashed as "LS100"'s )(rear monitor), Those all work perfectly fine as intended.

The LT's just exhibit this strange behavior. It isnt limited to any specific mode (both ambient screen and any configured colour mode) all do it (ie if they are full white, random LED's will flicker to black back to white or change to say blue or orange and back to white)

I've tried using a different fresh Arduino aswell as repurposed the "LS100" arduino (since i knew it worked), same issue.
Replaced the USB cable, used it on a different port, same issue.
Reflashed it, made sure it had its own identifier in the firmware, same issue.
Pulled the Light tower apart and triple checked all my wiring of the LED strips. Zero solder bridges or bad connections anywhere.
Checked voltages from my power supply (which powers under desk/monitor LED's too), no issues there either.

Am i missing something? Is it a icue or firmware issue? I am running it with the "increased brightness" flag in the firmware, could that be causing an issue? (i havn't tried flashing that out yet)

Anyway direction here would be great. 

Thanks

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Pro Micro
 - Device: LT100
 - Version of IDE: Arduino IDE 1.8.13
 - Version of CLP: 0.15.0
 - Version of CLP Boards: unknown
 - Sketch: LT100

**Code changes**
SerialNumber
Fixicuebrightness

Legion2, 2022-11-01:
This sounds like a faulty LED, have you checked with a different LED strip if the problem still persists?

GodlyOverlord, 2022-11-03:
Well it does it on the 2 different towers. The towers also have 4 "single led" strips in the bottom that are wired in parallel (so all 4 show the same output for the bottom section of the LT100) and all 4 single led strips often flicker in unison (aswell as random LED's on the main tower strip). 

Its entirely possible its one of the LED's causing the issue (maybe distorting the signal somewhere?). I thought maybe the link cable between the 2 towers may of introduced something but when i disconnect the second tower the first still flickers randomly/doesn't change

I tried reflashing the arduino without the icue brightness fix, it seemed good for a night but tonight its going crazy again. I guess if its not a firmware issue it has to be hardware, and i've its not power or the arduino that really only leaves the LED strip right?

Legion2, 2022-11-04:
Are the GND of the LEDs connected to GND of the Arduino? Can you draw a schematic of your wiring?

closed 2022-11-27 by github-actions[bot]