Legion2/CorsairLightingProtocol #313: FPS counter on mega-uno boards

sudousercom, 2022-11-20:
I added a FPS counter in #176. Look at the examples how it is used https://github.com/Legion2/CorsairLightingProtocol/pull/176/files. You need to download the dev version of the library to use the FPS counter.

For me it prints 24/25 fps which is totally smooth.

_Originally posted by @Legion2 in https://github.com/Legion2/CorsairLightingProtocol/issues/175#issuecomment-683592481_

Hi! How can i use fps counter on mega board with hoodloader2?

Legion2, 2022-11-22:
The fps counter prints the fps to serial monitor, however on mega board with hoodloader2 the serial monitor is used for USB communication, so it is not possible to use the normal serial monitor for this.

sudousercom, 2022-11-22:
> The fps counter prints the fps to serial monitor, however on mega board with hoodloader2 the serial monitor is used for USB communication, so it is not possible to use the normal serial monitor for this.

thank you for explanation

closed 2022-11-27 by Legion2