Legion2/CorsairLightingProtocol #315: Compilation error: No connection established (Corsair Lighting Protocol)

Agademo, 2022-11-24:
**Compilation error: No connection established**
Hello everyone, I followed all the steps on the Corsair Lighting Protocol page in order, but when I click on the print library tab, I get the error "Compilation error: No connection established". Can you help me please? Thank you.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - Device: [e.g. Lighting Node PRO]
 - Version of IDE: [e.g. Arduino IDE 2.0.0 or 2.0.2]
 - Version of CLP: [e.g. 0.15.0]
 - Version of CLP Boards: [e.g. 0.2.0]
 - Sketch: [e.g. RepeatAndScale]

**Code changes**
i didn't change

Legion2, 2022-11-24:
Hi, can you try to update Arduino IDE to 2.0.2 or later, there was a [bug](https://github.com/arduino/arduino-cli/issues/1970) which causes this error.

Agademo, 2022-11-24:
I got the same error in Arduino IDE 2.0.2

Legion2, 2022-11-24:
> but when I click on the print library tab
what do you mean with the "print library tab", can you share a screenshot please

Agademo, 2022-11-25:
I solved the problem. The problem is that I have uploaded the CLP code of a different board to the Arduino Leonardo board. Thanks for help. Now I have a different question. I can now burn a total of 60 pixels with the card in my hand. Is it possible to increase this number in the code?

Legion2, 2022-11-27:
Yes you can increase the number of pixels, there are two ways, 
1. choose a different example sketch, like the [LS100](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/LS100/LS100.ino), which allows you in iCUE to select different LEDs with more pixels.
1. Scale up the 60 pixel to e.g.120 pixels with the `scale` function, see https://github.com/Legion2/CorsairLightingProtocol#repeat-or-scale-led-channels

You can also combine both.

Keep in mind there is only limited memory available and each pixel needs memory, so in total you can get around 400 pixels.

Agademo, 2022-11-27:
Thank you so much for your reply and help man. Can we burn 400 pixels for each channel, or can we burn 400 pixels in total for both channels. In fact, if you had written an explanatory video, it would have been inedible :).

Agademo, 2022-11-27:
I wanted to share a picture to be more descriptive. There is a total of 10 meters of WS2812b strip LEDs on the edges of my table, which you can see. These LEDs have a total of 600 pixels. The shelf you see at the top has a total of 4 meters and 240 pixels. In this case, do you need two arduino leonardos separately for the table and shelf?
[image]

Legion2, 2022-11-27:
You can get around 400 pixel total with an arduino leonardo, I recommend using one channel of the LS100. The LS100 provides ambient lighting functions in iCUE. You can configure 135 pixels in iCUE and scale them up to around 400 pixel with the arduino (you should test/try which is the maximum, before building something).
Are all your 600 pixels individually addressable, or are they in groups of 3 or so?

Agademo, 2022-11-27:
I divided all 600 pixels into 4 different groups.

closed 2022-12-20 by github-actions[bot]

Agademo, 2023-02-12:
Hello, how exactly can I increase the number of pixels, what values ​​should I enter? can you help? Total number of addressable pixels 600

Legion2, 2023-02-13:
Which of the both approaches do you want from https://github.com/Legion2/CorsairLightingProtocol/issues/315#issuecomment-1328255374 ?

Agademo, 2023-02-13:
Ekteki resimde de göstermiş olduğum gibi Arduino Leonardo ile LS100 kontrolcü oluşturdum. Bu kontrolcü ile şuan 135 piksel led yanıyor. Daha önce bu kod ile maksimum 400 piksel led yanabileceğini söylemiştiniz. LS100 için hazırlanmış kodda 
CRGB ledsChannel1[135];
CRGB ledsChannel2[54];
şeklinde değerler mevcut. Maksimum 400 piksel led yakmam için kodu nasıl ayarlamam gerekiyor?

[image]

Agademo, 2023-02-14:
Very happy if you help🙏

Legion2, 2023-02-14:
see #325