Legion2/CorsairLightingProtocol #316: LS100 Question

Mrjose77, 2022-11-28:
I builded a LS100 the last week, actually I'm using two Chanels on it, one is for ambient display of the monitor and the second one is for an external led strip

The problem here is when I shutdown the computer, the leds of the second Channel (the external led strip) didn't turn off, they just freeze, so I have to turn on the computer and press the button (the ls100 one) to power off the led strip and then turn off the pc.

So I would like to ask is this is normal? Or there is way to setup it to go off when the computer is shutting down ?

I'm using a Rasperry Pi Pico whit a logic level shifter, im running 84 leds on channel one and 69 on channel two

This is the diagram that I'm using 

[image]

Legion2, 2022-11-28:
Have you configured Hardware lighting effects in iCUE for the LS100?

Mrjose77, 2022-11-29:
Yes i have, i'm using black color in ICUE for the Hardware lighting in the 2 channels like in the picture

[image]

Legion2, 2022-11-29:
Can you share the code you use for the LS100?

Mrjose77, 2022-11-29:
This is the code

```
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

#define BUTTON_PIN 4

CRGB ledsChannel1[84];
CRGB ledsChannel2[84];

// The Serial Number can be set in the constructor
const char mySerialNumber[] PROGMEM = "202B6949A977";

// Most ARM devices do not contain an EEPROM; we will use static storage for the Device ID
DeviceID deviceID = {0x8A, 0xDA, 0xA8, 0x8E};
CorsairLightingFirmwareStorageStatic firmwareStorage(deviceID);
CorsairLightingFirmware firmware(CORSAIR_SMART_LIGHTING_CONTROLLER, &firmwareStorage);
FastLEDController ledController(nullptr);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolTinyUSBHID cHID(&cLP, mySerialNumber);

void setup() {
	cHID.setup();
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 84);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 84);
	ledController.addLEDs(0, ledsChannel1, 84);
	ledController.addLEDs(1, ledsChannel2, 84);
	pinMode(BUTTON_PIN, INPUT_PULLUP);  
}

void loop() {
static bool lightingEnabled = true;
	cHID.update();

	if (buttonClicked()) {
		lightingEnabled = !lightingEnabled;
		fill_solid(ledsChannel1, 84, CRGB::Black);
		fill_solid(ledsChannel2, 84, CRGB::Black);
		FastLED.show();
	}

	if (lightingEnabled && ledController.updateLEDs()) {
		FastLED.show();
	}
}

/**
 * Handle button of the LS100. The button is optional.
 *
 * @return true if the button was pressed and then released.
 */

bool buttonClicked() {
	static bool previousState = 1;
	bool state = digitalRead(BUTTON_PIN);
	if (previousState == 0 && state == 1) {
		previousState = state;
		return true;
	}
	previousState = state;
	return false;
}
```

Legion2, 2022-12-04:
When the computer is turned off the LEDs should go into hardware mode after 30 seconds. Also the button should also work if the computer is turned off. So you don't have to restart the computer to turn off the LEDs with the button.

Mrjose77, 2022-12-07:
I finally found the problem, the circuit and the code were fine, the issue was on the computer specified in the USB ports because I'm using an S4+S5 power state to turn off all the USB devices in the computer when it is off, so I disable this option and seems to be working fine.

closed 2022-12-07 by Mrjose77