Legion2/CorsairLightingProtocol #322: Arduino Leonardo and Pro Micro not working properly with CorsairLightingProtocol

mesoruco, 2023-01-20:
Since the beginning of January, my arduino Leonardo stopped working correctly with the CorsairLightingProtocol, first I thought it was a SignalRGB update issue but it also failed on ICUE software, then thought it was a board error and bought an arduino pro micro but the operation, or malfunction, is the same. I tried to upload the protocol from another computer thinking it was an arduino software problem but the malfunction continues

**Screenshots**
Don't have pictures, but the idea is that the RGB fans should light like rainbow but just flashes white with some random colors occasionally. Tested the fans with a Razer Chroma RGB addressable controler and they illuminate just fine with the right colors

 - OS: Windows 10
 - Board: Arduino Leonardo and Arduino Pro Micro
 - Device: Lighting Node PRO
 - Version of IDE: Arduino IDE 2.0.3
 - Version of CLP: 0.15.0
 - Version of CLP Boards: 0.3.1
 - Sketch: LightingNodePro

Did you changed the code of the library or the sketch? NO, didn't change anything

Legion2, 2023-01-20:
This sounds like one of the LEDs is broken and causes all subsequent LEDs to flash in random colors. Can you test with another LED strip/fan to check if a defect LED causes these problems?

mesoruco, 2023-01-21:
> This sounds like one of the LEDs is broken and causes all subsequent LEDs to flash in random colors. Can you test with another LED strip/fan to check if a defect LED causes these problems?

I have tested both arduinos (Leonardo and Pro Micro) with five different argb fans and three different led strips without luck.
Also, right now I have these fans and strips connected to two razer chroma addressable controllers and all work properly, showing correct lighting patern using SignalRGB

Legion2, 2023-01-21:
How did you connect the power supply to the Arduino and LEDs? Do they have a common Ground?

mesoruco, 2023-01-21:
> How did you connect the power supply to the Arduino and LEDs? Do they have a common Ground?

Yes, I'm using a molex connector, all grounds from arduino and leds goes to it

Legion2, 2023-01-21:
Is the data wire soldered to the leds and the Arduino? Maybe there is a loose wire?

mesoruco, 2023-01-21:
> Is the data wire soldered to the leds and the Arduino? Maybe there is a loose wire?

Yes, every cable is soldered in both arduino boards. As I mentioned when posted, the Arduino Leonardo worked just fine for about four months with SignalRGB, then at the begining of January, it simply stopped working, then I tried to use the Arduino Pro Micro and from the begining, it didn't work well, now, both the arduino Leonardo and the Arduino Pro Micro behave as mentioned

mesoruco, 2023-01-23:
Hi there, managed to buy a second Arduino Pro Micro, sodered every cable n doublechecked them and this time, this one seems to be working just fine. Also discovered thar SignalRBG added an option to optimize arduino performance for the corsair lighting node on its app, so everything appears to be ok as for now. Guess it is a problem with the other arduino boards? I'll take them to the store to check for errors, although they both work fine with other sketchs, so there's some mistery around this story xD anyway thxs a lot for your time n effort to try to fix this strange issue, regards

closed 2023-01-23 by Legion2