Legion2/CorsairLightingProtocol #326: Changing serial number and id with Devicetool

Agademo, 2023-02-17:
Hello, I had previously created a Corsair Smart Lighting LS100 with Arduino Leonardo and increased the number of leds to 400. Toplamda 600 piksel lede sahip bir masam ve 240 piksel lede sahip bir raf var.. The leds on my Desk and shelf are ws2812b. There are a total of 600 pixel LEDs on my desk and a total of 240 pixel LEDs on the shelf. For this reason, I want to write LS100 code on two separate arduino leonardos for my desk and an arduino leonardo for the shelf and run all the pixels. But I need to change the serial number and id number of each leonardo. Since I am not very knowledgeable about arduino, I need to set different serial number and id number for each arduino via devicetool. I would be very grateful if you could help me explain this. Thank you for your help.

 - OS: [e.g. Windows 11]
 - Board: [e.g. Arduino Leonardo]
 - Device: [e.g. LS100]
 - Version of IDE: [e.g. Arduino IDE 1.8.19]
 - Version of CLP: [e.g. 0.15.0]
 - Version of CLP Boards: [e.g. 0.2.0]
 - Sketch: [e.g. Devicetool]

Legion2, 2023-02-17:
You need to create two sketches, one for each of your arduinos. Then add different serial number and device id as described [here](https://github.com/Legion2/CorsairLightingProtocol#use-of-multiple-devices) to both of them.

Agademo, 2023-02-17:
How do I create the sketch?

Legion2, 2023-02-17:
Copy the code from #325 and create new sketches in Arduino IDE with the code via the menu. Each sketch is a file with the `.ino` file ending.

closed 2023-03-12 by github-actions[bot]