Legion2/CorsairLightingProtocol #328: commander pro problem

Black-Sparro, 2023-02-22:
good afternoon dear legion2.
thanks for you're hard work on this project. i test it and love it. it works great. however i find myself a problem that i can not debug. every sketch i use and upload, ICue software registers it as lightning node pro. i want to try commander pro but is no use. maybe i have to change device id or PID somewhere but i couldn't find any guide in git-hub or comment on code to solve it. my hardware is Arduino pro micro. needless to say with commander pro sketch, still i can use 2 ARGB channels.

Legion2, 2023-02-22:
Hi, thanks, in the Tools menu you need to select the correct "Device" type you like to create:

[image]

Black-Sparro, 2023-02-22:
thanks a lot man.

Black-Sparro, 2023-02-22:
works like a charm

closed 2023-02-22 by Legion2