Legion2/CorsairLightingProtocol #330: Correct iCue configuration for MultiFans

SanjeetPathak, 2023-02-28:
HI @Legion2,

First of all many thanks to the project. With the awesome project and detailed guide I was able to software controlled the lighting of my PC. While everything works flawless, I'm facing challenge in setting up MultiFans.

Current ARGB Fan config:

Total 7 Fans are connected to a single channel, out of which

1 Fan is 5 Led
3 Fans are 6 Leds
3 Fans are 16 Leds

The other channel is being used for ARGB strip.

Now coming to the issue, I want fan ARGB to be controlled independently. Once possible solution could be adding more Arduinos. I have several Arduinos lying around with the problem with this is:

1. Limited internal USB header, in my case just 1 which I'm already using.
2. increasing number of wire clutters.

I found that you have added MultiFans.ino also which seems to flash correctly and I could see that I can modify the number of LEDs per fan also that I already did and flash. I'm now clueless how should I set that up in iCue. iCue doesn't have anything for uneven LEDs. 

Tried searching this in issue section and in documentations but was not able to find. Help would highly be appriciated.

With ❤️ from India!

Legion2, 2023-02-28:
Hi,
for you limited internal USB headers, you can use NZXT Internal USB Hub, I'm using it too.
For the different fan sizes, you can combine the MultiFans.ino with the [segment scaling feature](https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel), it allows you to map iCUE fan leds to custom fan leds. It does this, by defining segments (iCue fans) of different sizes and allows them to be remapped to segments (your fans) of different sizes. I hope you are familiar with programming, else this is very complicated to setup.
I hope this helps you.

SanjeetPathak, 2023-02-28:
@Legion2 Thanks for the response. Yes I'm also a developer. Not too much in hardware stuffs but was able to understand by your awesome documentation. I was working just on the code and was actually able to use the repeat function to make an even lighting setup in uneven fans. Also I chose 8 LED series fans in setup. Will combine it to MultipleFan.ino.

closed 2023-03-23 by github-actions[bot]