Legion2/CorsairLightingProtocol #331: Is it possible or it will be in the future ST100?

SkyBadPT, 2023-03-14:
Hi,

This is more like a question, not a problem, I would like to know if it is possible to emulate the ST100 or if it will be in the future. I could not find any information about it on the page.

Thanks!

Legion2, 2023-03-15:
I think the ST100 is similar to the LT100, so it should be possible to support it. We need to know the USB HID ID of an original ST100 device, so we can replicate it.

SkyBadPT, 2023-04-08:
> I think the ST100 is similar to the LT100, so it should be possible to support it. We need to know the USB HID ID of an original ST100 device, so we can replicate it.

Do you know how can I get that ID?

Legion2, 2023-04-09:
The USB PID can be obtained from the original device or by guessing it. If you have an original device you can use windows device manager and look for the USB device, in the details page of the device, there are multiple attributes, one of it contains the USB HID ID.

Mikazil, 2024-01-23:
> The USB PID can be obtained from the original device or by guessing it. If you have an original device you can use windows device manager and look for the USB device, in the details page of the device, there are multiple attributes, one of it contains the USB HID ID.

Hi, I found Pid and vid.
PID - 0A34
VID - 1B1C

SkyBadPT, 2024-09-28:
Can someone check if the information that Mikazil shared can be implemented? I would thest it if I could but I have no clue on how to implement it :(