Legion2/CorsairLightingProtocol #338: Is there a 256 leds limit?

HasseM, 2023-06-02:
I'm trying to output to more then 256 leds but when I make SIZE in ledsChannel0[SIZE] greater dan 255, the microcontroller seems to crash (blinking led not blinking). At Size = 255 it all works, as soon as I make it 256, it doesn't.

Using an RPi Pico as a Lightning Node Pro. Pico is very capable and sketch is just 3% of storage capacity and memory is at 4%. so that's not the problem. Also checked fastLed lib for limits, but fastLed does not have a limit to the amount of leds.

Any advise where to start looking?

HasseM, 2023-06-06:
Still there seems to be an issue with more then 255 leds. I have 806 leds and only 38 are received from iCue.
806 = 0011 0010 0110
 38 =           0010 0110

It seems that they are not copied from usb to ledChannel0[] array. So probably an uint8_t there somewhere.

Legion2, 2023-06-06:
Which type of hardware do you use, normally the memory only allows for a total of around 400 LEDs. You can not add that much LEDs via `addLeds` because iCUE does not support that much leds. You can add for example 135 LEDs, which maps to the maximum which is possible with iCUE LS100 Controller. Then you can scale that to a bigger LED array with the scale functions see https://github.com/Legion2/CorsairLightingProtocol#repeat-or-scale-led-channels

closed 2023-06-21 by Legion2