Legion2/CorsairLightingProtocol #344: Pro micro, cant run argb strip

slayernn021, 2023-07-12:
Hello, im using Arduino Pro Micro ATmega32U4 16Mhz blue board (clone). Argb fans work perfectly in both outputs, controling them true icue. Problem comes when i put argb strip, it wont power up, tried both outs and different examples(like neopixel etc etc). Tried same strip on arduino mega,work fine. Complete mistery to me. Any suggestion?

Legion2, 2023-07-12:
Hi, I don't understand, what do you mean with `both outputs`? Can you please make a picture or schematic of the wiring you are using to connect the argb strip and arduino.

slayernn021, 2023-07-12:
Hi i mean output in terms of 2 connectors for ie argb fan and strip. Out = digital pin 2 and pin 3 on board. 
Im using standard wiring, 470ohm resistor on D-pin and 1000uf cap on power. ARGB Fans work perfect on both pin 2/3. Strip no response, i checked with multimer, it gets 5v. Same wiring on arduino mega, same strip = working. I dont understand

Legion2, 2023-07-12:
What is the exact type of the led strip, you have to configure the type in `FastLED.addLeds` call, so it can control the led strip. Do you use an external power supply for the strip?

slayernn021, 2023-07-12:
Its WS2811 strip , RGB+IC 14W 5V 60 led/meter.
So i tried to change WS2812b to WS2811 in default example:
FastLED.addLeds<WS2812b, DATA_PIN_CHANNEL_1, RGB>(ledsChannel1, 96);
FastLED.addLeds<WS2812b, DATA_PIN_CHANNEL_2, RGB>(ledsChannel2, 96);
Still not working. What else to try? Sorry im not that skilled in programing, just know some basics.

Legion2, 2023-07-12:
have you connected ground of the strip to the arduino and the external power (common ground)?

slayernn021, 2023-07-12:
I did exacly like on image. All gnd (from 2 ARGB plugs) are wired to pro micro gnd , - on capacitor, gnd from PSU molex. +5v from molex goes to 2 ARGB plugs. Data pin 2 and 3 are connected to ARGB plugs via 470ohm. Mistery?
[image]

Legion2, 2023-07-12:
Have you tested the led strip hardware by using a simple FastLED test sketch to just test that the strip works?

slayernn021, 2023-07-13:
I tried now with FastLED sketches, not working... i rly dont get. Strip is 100% good, everything else too. This pro micro clone is smelly to me, but then how argb fans are working normal :\ So weird.

slayernn021, 2023-07-13:
I will add another stripe of 20 leds which will be total of 60 like in beggining and then i will cut one by one led from strip so i come to max leds i can fire up from my pro micro crap