Legion2/CorsairLightingProtocol #345: Use of PWMFan causes device to disconnect/freeze when setting RGB on Arduino Pro Micro clone

xCuri0, 2023-07-12:
**Describe the bug**
Even just declaring a `PWMFan` causes my Arduino Pro Micro to disconnect or freeze when RGB is set. I also tried commenting out the timer related code in it and it made no difference.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Commander Pro Sketch and upload it
2. Set RGB, I'm using OpenRGB haven't tried iCUE
3. Device will either disconnect or stop responding to RGB values

**Expected behavior**
I expect it to not crash when setting RGB, if I use the Lighting Node Pro sketch or remove `PWMFan` from Commander Pro it works fine. Even without the use of `addFan` and/or `updateFans` it will cause this behaviour.

**System (please complete the following information):**
 - OS: Windows 10 22H2
 - Board: Arduino Pro Micro clone
 - Device: Commander Pro
 - Version of IDE: Arduino IDE 1.8.19
 - Version of CLP: 0.15.0
 - Version of CLP Boards: 0.3.1
 - Sketch: Commander Pro

I currently only have an ARGB fan connected but I do plan on adding a 4 pin fan once I figure this out

**Code changes**
None, I have to modify Commander Pro to remove `PWMFan` if I want to use it without crashing.

Legion2, 2023-07-12:
Calling the `PWMFan` constructor configures timer registers of the arduino to support higher refresh rates. However these settings has only been tested on arduino mega. Maybe you can check if these settings are correct or need to be changed for Arduino Pro Micro, so we can add support for it.

https://github.com/Legion2/CorsairLightingProtocol/blob/c0f4fd1e29a8011a55153514e37cd5633465df35/examples/CommanderPRO/PWMFan.cpp#L21-L46

xCuri0, 2023-07-13:
So if I set OpenRGB to only use Channel 1 it works fine using unmodified Commander Pro example, is this a memory issue or something similar ?

closed 2023-07-13 by xCuri0

xCuri0, 2023-07-15:
even 88 might be too high, 48 seems safe and 56 may work