Legion2/CorsairLightingProtocol #346: Error: invalid offset, value too big (0x0000049C) compiling for pi pico

haze1986, 2023-08-14:
**Describe the problem**
I have previously successfully compiled and uploaded for a pi pico before. After a few months and getting a new pi pico I am having issues with compilation

**Screenshots**
Error code when compiling
`
In file included from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/led_sysdefs.h:35,
                 from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/FastLED.h:51,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/FastLEDController.h:18,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/CorsairLightingProtocol.h:34,
                 from C:\Users\Haze\AppData\Local\Temp\.arduinoIDE-unsaved2023714-2624-rg4qa7.6fgs\TinyUSB\TinyUSB.ino:16:
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h:87: warning: "cli" redefined
   87 | #define cli() (saved_interrupt_status = save_and_disable_interrupts())
      | 
In file included from C:\Users\Haze\AppData\Local\Temp\arduino\sketches\8B6D884B987E9C922631E15A455016CF\sketch\TinyUSB.ino.cpp:1:
C:\Users\Haze\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.3.2\cores\rp2040/Arduino.h:74: note: this is the location of the previous definition
   74 | #define cli() noInterrupts()
      | 
In file included from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/led_sysdefs.h:35,
                 from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/FastLED.h:51,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/FastLEDController.h:18,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/CorsairLightingProtocol.h:34,
                 from C:\Users\Haze\AppData\Local\Temp\.arduinoIDE-unsaved2023714-2624-rg4qa7.6fgs\TinyUSB\TinyUSB.ino:16:
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h:88: warning: "sei" redefined
   88 | #define sei() (restore_interrupts(saved_interrupt_status))
      | 
In file included from C:\Users\Haze\AppData\Local\Temp\arduino\sketches\8B6D884B987E9C922631E15A455016CF\sketch\TinyUSB.ino.cpp:1:
C:\Users\Haze\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.3.2\cores\rp2040/Arduino.h:73: note: this is the location of the previous definition
   73 | #define sei() interrupts()
      | 
In file included from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastled_arm_rp2040.h:5,
                 from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms.h:35,
                 from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/FastLED.h:62,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/FastLEDController.h:18,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/CorsairLightingProtocol.h:34,
                 from C:\Users\Haze\AppData\Local\Temp\.arduinoIDE-unsaved2023714-2624-rg4qa7.6fgs\TinyUSB\TinyUSB.ino:16:
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:28:42: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   28 |   inline static void set(register port_t val) __attribute__ ((always_inline)) { sio_hw->gpio_out = val; }
      |                                          ^~~
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:34:45: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   34 |   inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { hi(); }
      |                                             ^~~~
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:35:45: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   35 |   inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { lo(); }
      |                                             ^~~~
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:36:50: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   36 |   inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port = val; }
      |                                                  ^~~~
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastpin_arm_rp2040.h:36:72: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   36 |   inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port = val; }
      |                                                                        ^~~
In file included from c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/FastLEDController.h:18,
                 from c:\Users\Haze\Documents\Arduino\libraries\CorsairLightingProtocol-0.15.0\src/CorsairLightingProtocol.h:34,
                 from C:\Users\Haze\AppData\Local\Temp\.arduinoIDE-unsaved2023714-2624-rg4qa7.6fgs\TinyUSB\TinyUSB.ino:16:
c:\Users\Haze\Documents\Arduino\libraries\FastLED\src/fastspi.h:172:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
  172 | #      pragma message "Forcing software SPI - no hardware SPI for you!"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:873: Error: invalid offset, value too big (0x0000049C)

exit status 1

Compilation error: exit status 1
`

**System (please complete the following information):**
 - OS: [Windows 10]
 - Board: [Pi Pico ]
 - Device: [Lighting Node PRO]
 - Version of IDE: [arduino-ide_2.0.3_Windows_64bit and arduino-ide_2.1.1_Windows_64bit]
 - Version of CLP: [CorsairLightingProtocol-0.15.0]
 - Version of CLP Boards: [Not sure]
 - Sketch: [TinyUSB]

**Code changes**
Did you changed the code of the library or the sketch?
No

sduponch, 2023-09-01:
I have a similar compilation error, the code build only when comenting out the line 37 to use only one Fastled.

haze1986, 2023-09-03:
> I have a similar compilation error, the code build only when comenting out the line 37 to use only one Fastled channel.

Hello, yes thanks. I verified that using just one fastled channel works. Will look forward to the full solution.

wandersonpassos, 2023-09-07:
I managed to compile with the 2 channels, just comment line 87 and 88 of the file below
libraries\FastLED\src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h

see if you can fix your problem

sduponch, 2023-09-18:
Ok, it is working fine.

sduponch, 2023-09-19:
In fact only the last channel is working now

wandersonpassos, 2023-09-19:
Did you define the number of LEDs on the channel? At first, when I managed to compile with 2 channels, I came across this, I thought it hadn't worked, I use OpenRGB, then I remembered that I had to define the number of channels, I currently use the project in a case where there is an LED strip on the front controlled by channel 1 and fans controlled by channel 2... check the number of LEDs on the channel

sduponch, 2023-09-19:
I tested with the following code :
`#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[245];
CRGB ledsChannel2[245];

// Most ARM devices do not contain an EEPROM; we will use static storage for the Device ID
DeviceID deviceID = {0x9A, 0xDA, 0xA7, 0x8E};
CorsairLightingFirmwareStorageStatic firmwareStorage(deviceID);
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDController ledController(nullptr);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolTinyUSBHID cHID(&cLP);

void setup() {
	cHID.setup();

	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 245);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 245);
	ledController.addLEDs(0, ledsChannel1, 204);
        ledController.addLEDs(1, ledsChannel2, 204);
        ledController.onUpdateHook(0, []() {
		CLP::scale(&ledController, 0, 245);
	});
        ledController.onUpdateHook(1, []() {
		CLP::scale(&ledController, 1, 245);
	});
}

void loop() {
	cHID.update();
	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}`

I ended up on wirerring my strips on the same 2nd channel as it is not critical at this moment but soon i will need the both channels.

Mrjose77, 2023-10-01:
I faced the same error, check this https://github.com/Legion2/CorsairLightingProtocol/issues/340

devastar, 2023-10-03:
> I faced the same error, check this #340

Didn't work for me.

Could anyone please confirm working versions of FastUSB, Pico library etc

closed 2023-10-26 by github-actions[bot]