Legion2/CorsairLightingProtocol #353: more than 2 channels

saeedbs, 2023-10-18:
is it possible to have more than 2 channels? 
or somehow have 2 corsair virtual device on 1 Arduino?  something like lighting node pro and commander pro on 1 arduino at same time  so we get 2 channel from each

Legion2, 2023-10-18:
No, but you can split a channel to use multiple led devices on one channel, see https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino

saeedbs, 2023-10-23:
I know that but more channels could be nice

Legion2, 2023-10-23:
iCUE only supports two channels.

closed 2023-10-24 by saeedbs

saeedbs, 2023-11-04:
what if I use openrgb and not icue ? is it possible to define more channel?

Legion2, 2023-11-04:
The number of channels are defined by the device types, and Corsair hat only defined device types with up to two channels. There is no way to tell openrgb the number of available channels via the USB protocol.