Legion2/CorsairLightingProtocol #354: Cannot compile for Arduino Uno 16u2

cschlisner, 2023-11-03:
**I have followed the guide exactly and cannot compile the project in IDE v2.2.1 or IDE v1.8.18**

I have confirmed Hoodloader2 flashed correctly:
[image]

And I am now trying to follow the install guide here: 
https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-on-Arduino-Uno-and-Arduino-Mega#upload-the-sketches

And I am getting errors related to HID:
[image]

Even though I have the HID-Project library installed. 

I also was getting a similar error when I tried it in an older IDE Version (1.8.18):
[image]

Is this even possible?

cschlisner, 2023-11-03:
I managed to get it to compile and upload by just leaving the board as "arduino Uno" but I doubt that will actually work. I tried proceeding but Step 8 of the installation guide says "select Tools->Device->Lighting Node PRO" 
However this menu does not exist:
[image]

I got this error when attempting step 10. 
[image]

Legion2, 2023-11-20:
> However this menu does not exist:
Is this screenshot from Arduino IDE 2 or 1? Please only use Arduino IDE 1, I have not tested 2 jet.

closed 2023-12-13 by github-actions[bot]