Legion2/CorsairLightingProtocol #355: commander pro with 1 analog channel and 1 digital channel

saeedbs, 2023-11-10:
I built a 'commander pro'  with Leonardo and 2 channels but one is digital and other one is analog, 
I mixed two of examples for this : nonaddressable and commander pro and removed some of fan controller codes
but most of the time my addressable rgb corsair fans don't work correctly while analog rgb fans work just fine, I think maybe I was wrong somewhere in code pls help me about it

Uploading video_2023-11-10_00-53-53.mp4…
[image]
[image]

```c++
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#include "SimpleFanController.h"
#include "ThermistorTemperatureController.h"

#define DATA_PIN_CHANNEL_1 2

//for channel 2
#define RED_PIN 9
#define GREEN_PIN 10
#define BLUE_PIN 11

#define TEMP_SENSOR_PIN_1 A4
#define TEMP_SENSOR_PIN_2 A5

#define FAN_UPDATE_RATE 500

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_COMMANDER_PRO, &firmwareStorage);
ThermistorTemperatureController temperatureController;
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
SimpleFanController fanController(&temperatureController, FAN_UPDATE_RATE, EEPROM_ADDRESS + storage.getEEPROMSize());
CorsairLightingProtocolController cLP(&ledController, &temperatureController, &fanController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

CRGB ledsChannel1[96];
CRGB ledsChannel2[10];

void setup() {
         CLP::disableBuildInLEDs();
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 96);
         ledController.addLEDs(0, ledsChannel1, 96);

        pinMode(RED_PIN, OUTPUT);
	pinMode(GREEN_PIN, OUTPUT);
	pinMode(BLUE_PIN, OUTPUT);
  ledController.addLEDs(1, ledsChannel2, 10);
	ledController.onUpdateHook(0, []() {
		// use color of first LED of the first channel
		set4PinLEDs(ledsChannel2[0]);
	});
        temperatureController.addSensor(0, TEMP_SENSOR_PIN_1);
	temperatureController.addSensor(1, TEMP_SENSOR_PIN_2);
}

void loop() {
	cHID.update();
	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

void set4PinLEDs(const CRGB& color) {
        analogWrite(RED_PIN, color.r);
	analogWrite(GREEN_PIN, color.g);
	analogWrite(BLUE_PIN, color.b);

}
```

saeedbs, 2023-11-10:
video their odd behavior
https://github.com/Legion2/CorsairLightingProtocol/assets/9435467/8e22f233-1ab7-41f0-9b5d-d5dd03f8b5db

Legion2, 2023-11-18:
Hi, your schematic and code looks good. Most often flickering comes from not having a common ground, bad wire connections or faulty LEDs.

closed 2023-12-10 by github-actions[bot]

saeedbs, 2023-12-20:
it seems leonardo is faulty ... after replacing it ...it works with no issue