Legion2/CorsairLightingProtocol #360: Reverse segments inside a channel

tiburcillo, 2024-01-22:
**Describe the problem**
I need to mirror reversely wired LED strips and I'm struggling with accomplishing this with my limited programming knowledge. I don't know if it is even possible or if I need to rip out the led strips.
The case has limited room for wires so it would be nice if I can keep the current wiring and solve the issue on software level.
How do I reverse segments of the top strip and the inside strip, but let the outside strip run normally?

**Screenshots**
[image]

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Sparkfun Pro Micro
 - Device: Lighting Node PRO
 - Version of IDE:Arduino IDE 1.8.13
 - Version of CLP: 0.15.0
 - Version of CLP Boards:0.2.0
 - Sketch: see below

**Code changes**
`/*
   Copyright 2020 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_FAN 16

// actually used
#define DATA_PIN_FAN_1 7
// TOP strip
#define DATA_PIN_STRIP_1 8
// FRONT OUTSIDE strip
#define DATA_PIN_STRIP_2 9
// FRONT IN strip
#define DATA_PIN_STRIP_3 10

CRGB ledsChannel1[21];
CRGB ledsChannel2[50];

const char mySerialNumber[] PROGMEM = "xxxxxxxx";

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  CLP::disableBuildInLEDs();
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
    
	// strip + case panel + external
	//                          FAN_PIN           CRGB array            offset             number of leds per fan
  // INVERT Green/red for SignalRGB
	FastLED.addLeds<WS2812B, DATA_PIN_STRIP_1, RGB>(ledsChannel2, 0, 30);
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_2, GRB>(ledsChannel2, 30, 50);
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_3, GRB>(ledsChannel2, 30, 50);
  
ledController.addLEDs(0, ledsChannel1, 21);
ledController.addLEDs(1, ledsChannel2, 50);
  ledController.onUpdateHook(1, []() {
    CLP::reverse(&ledController, 1);
  });

}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}
`

Legion2, 2024-01-22:
You can create a new LED buffer `CRGB ledsChannel2Reverese[20];` in which you copy the led values in reverse order and then hook this up to `FastLED.addLeds<WS2812B, DATA_PIN_STRIP_3, GRB>(ledsChannel2Reverse, 0, 20);` and before `FastLed.show();` insert this code to copy the values 
```c++
 for (uint8_t i=0; i<20; i++) {
    CRGB ledsChannel2Reverese[20 - 1 - i] = ledsChannel2[30 + i];
  }
```

tiburcillo, 2024-01-22:
Genius! It worked. I first was struggling with the placement of the for loop. I had to put it just before the FastLed.show(); function.
Will post a picture of the result.

Here's my (Leon's) working code in case somebody needs it (I also use the scale function to adapt my Chinese 18-led FAN to a Corsair HD fan with 12 leds):
`/*
   Copyright 2020 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

// spare
//#define DATA_PIN_FAN_3 7
//#define DATA_PIN_FAN_4 8
//#define DATA_PIN_FAN_5 9
//#define DATA_PIN_FAN_2 9
//#define DATA_PIN_STRIP_2 15

// actually used
#define DATA_PIN_FAN_1 7
#define DATA_PIN_STRIP_1 8
#define DATA_PIN_STRIP_2 9
#define DATA_PIN_STRIP_3 10

CRGB ledsChannel1[18];
CRGB ledsChannel2[50];
CRGB ledsChannel2Reverse[20];

const char mySerialNumber[] PROGMEM = "2024B0BB1002";

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  CLP::disableBuildInLEDs();
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, 0, 18);
      
	// strip + case panel + external
	//                          FAN_PIN           CRGB array            offset             number of leds per fan
  // INVERT Green/red for SignalRGB
  // TOP
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_1, GRB>(ledsChannel2, 0, 30);
  // INSIDE
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_3, GRB>(ledsChannel2, 30, 50);
  // FRONT
  FastLED.addLeds<WS2812B, DATA_PIN_STRIP_2, GRB>(ledsChannel2Reverse, 0, 20);
  
  ledController.addLEDs(0, ledsChannel1, 12);
	ledController.addLEDs(1, ledsChannel2, 50);
  ledController.onUpdateHook(0, []() {
    CLP::scale(&ledController, 0, 18);
  });
  //cfr sketch for explanation of double reverse for external led strip
  ledController.onUpdateHook(1, []() {
    CLP::reverse(&ledController, 1);
  });  
}

void loop() {
	cHID.update();
  for (uint8_t i=0; i<20; i++) {
    ledsChannel2Reverse[20 - 1 - i] = ledsChannel2[30 + i];
  }
	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}
`

closed 2024-01-22 by tiburcillo