Legion2/CorsairLightingProtocol #362: Multiple fans sketch not compiling for Arduino Uno

Lethaldroid, 2024-02-07:
**Describe the problem**
Multiple fans sketch not compiling for Arduino Uno. I followed all the steps and successfully made an corsair lighting node pro with 2 channels but when i try to upload a multiple fans one, it fails to compile the sketch for UNO saying that the sketch is too big

**System (please complete the following information):**
 - OS:  Windows 10
 - Board: Arduino UNO/ CLP Hoodloader 16u2
 - Device: Lighting Node PRO
 - Version of IDE: Arduino IDE 1.8.18
 - Version of CLP: 0.15.0
 - Version of CLP Boards: 0.3.1
 - Sketch: Multiplefans
**Code changes**
Did you changed the code of the library or the sketch?
- No

Lethaldroid, 2024-02-08:
`/*
   Copyright 2020 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_FAN 8

#define DATA_PIN_FAN_1 2
#define DATA_PIN_FAN_2 3
#define DATA_PIN_FAN_3 4
#define DATA_PIN_FAN_4 5
#define DATA_PIN_FAN_5 6
#define DATA_PIN_FAN_6 7
#define DATA_PIN_CHANNEL_2 8

CRGB ledsChannel1[96];
CRGB ledsChannel2[96];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolSerial cHID(&cLP);

void setup() {
	// 6 fans on channel 1
	//                          FAN_PIN           CRGB array            offset             number of leds per fan
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 3, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 4, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 5, NUMBER_OF_LEDS_PER_FAN);

	// normal strip on channel 2
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 96);
	ledController.addLEDs(0, ledsChannel1, 96);
	ledController.addLEDs(1, ledsChannel2, 96);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}`

This is the changed code. I changed the board to Hoodloader Uno. Now trying to upload the sketch to Arduino Uno as a CLP Hoodloader 16u2. My aim is to connect 10 fans to 10 digital pins of the arduino and control all of them individually.
This is the error im getting:
`In file included from C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/FastLED.h:75:0,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:34,
                 from C:\Users\GameBoy\AppData\Local\Temp\arduino_modified_sketch_174188\MultipleFans.ino:16:
C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
[... 15 lines cut ...]
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerStorage.h:18,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerStorageEEPROM.h:20,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerStorageEEPROM.cpp:16:
C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/FastLED.h:75:0,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDController.h:18,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerUtils.h:19,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerUtils.cpp:16:
C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\GameBoy\Documents\Arduino\libraries\FastLED\src/FastLED.h:75:0,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\LEDController.h:23,
                 from C:\Users\GameBoy\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src\LEDController.cpp:16:
[... cut ...]
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Sketch uses 27428 bytes (223%) of program storage space. Maximum is 12288 bytes.
text section exceeds available space in boardGlobal variables use 1673 bytes (326%) of dynamic memory, leaving -1161 bytes for local variables. Maximum is 512 bytes.

Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board CLP HoodLoader2 16u2.
`

Legion2, 2024-02-09:
Ok the code looks fine, but you have still selected the wrong Board, please check the Guide for more Info and step by step guide how to select the correct board

Lethaldroid, 2024-02-09:
I selected the board as Hoodloader 2 Uno and uploaded the MultipleFans sketch (after replacing `CorsairLightingProtocolHID` with `CorsairLightingProtocolSerial ` ) to the board which uploaded successfully.
Then i selected the board as CLP Hoodloader2 16u2, Device as LightingNodePro, the COM port and uploaded the Hoodloader2CLPbridge to the Arduino Uno and it seems to be uploaded now. I would now proceed to checking the board.

Lethaldroid, 2024-02-09:
the problem is that after compiling multiplefans sketch and uploading it to my arduino uno, the rgb data is not being sent through the digital pins. Instead some garbage data is being sent which makes the light strips erupt random colors and sometimes they just stay off

closed 2024-03-02 by github-actions[bot]