Legion2/CorsairLightingProtocol #363: Commander pro to 6 argb fans

JACKAR710, 2024-03-03:
I've been looking through your repository and it's really something. I was wondering if you have any advice on creating a controller that takes the data sent by a channel on the Commander pro, and converts it so that you'd be able to control multiple 3 pin argb devices. This would be very helpful when using signalRGB.

Kind regards,
A fellow embedded fanatic

Legion2, 2024-03-03:
What do you mean by 3 pin devices? Normal addressable led stripes? Have a look at https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples%2FMultipleFans%2FMultipleFans.ino. it controls multiple led strips per channel.

JACKAR710, 2024-03-03:
I read the code and I couldn't really figure out how the avr board is supposed to recieve data from the lighting node (or in my case the commander pro). With 3 pin argb I meant the goto argb connection.

Legion2, 2024-03-03:
What do you mean by 3 pin devices? Normal addressable led stripes?do you know how to use the FastLED library and how to wire the components?

closed 2024-03-25 by github-actions[bot]