Legion2/CorsairLightingProtocol #368: Hardware lightning not working

Fregoudi, 2024-04-16:
Hardware lighting not working. When I close the icue software or turn off my computer, the towers do not work and the LEDs remain solid. Can you help me?

 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - Device: [e.g. LT100]
 - Version of IDE: [e.g. Arduino IDE 1.8.57]
 - Version of CLP: [e.g. 0.15.0]
 - Version of CLP Boards: [e.g. 0.3.1]

Legion2, 2024-04-16:
You need to configure the hardware lighting in iCue before it works. Can you please provide a screenshot of your hardware lighting configuration in icue.

Fregoudi, 2024-04-16:
I remember doing the configuration before but I guess I didn't. I did what you said and it's working now. I had another question, how do I add a button for on/off operation?

Legion2, 2024-04-16:
The LS100 examples includes code for on/off button https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples%2FLS100%2FLS100.ino#L23

Fregoudi, 2024-04-16:
How do I add it to lt100?

Legion2, 2024-04-16:
You have to write the code yourself 😉

closed 2024-04-16 by Fregoudi