Legion2/CorsairLightingProtocol #370: Can I use 5 channels to control them independently of each other?

Artasdmc, 2024-04-17:
I've got an atmega 32u4 board that accepts up to 18v. So I have no issues with power, could just power it with 12v line on raw, and 5v line on vcc and another 5v low power input from internal motherboard connection. It has 5 PWM pins that I'd like to take advantage of.

In my system I have 6 fans with 8 leds each, and a IC strip roughly 160leds, making a ring in front of my meshify case, behind the mesh panel, tucked where you can't see the strip itself, but it should illuminate the front nicely.

Now since I only have 1 board,  I'd like to use it to it's fullest, make at least 5 channels that I could control individually. I've done it with fastled on my UNO board, but I want the ease of using it with OpenRGB, or corsair icue, without having to go through arduino IDE, coding, uploading and etc..

So is this limited just to 2 channels? 
Which would mean I'd have to get a second board to make a commander pro for fans separately?

Legion2, 2024-04-17:
I'm not sure how to power you board. But you can use all your pins to control your devices. There are only two icue channels, but each channel can be connected to multiple devices/pins. Have a look at this example on how to connect mulitple pins (fans) to a single channel https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino

Artasdmc, 2024-04-17:
Thank you.
I was just saying I'll have enough current to power more than what I have in my system.

Artasdmc, 2024-04-19:
Hi, I'm getting back to ask a question about scaling LED strip.
My strip has exactly 178 LEDs.

This is the code used to flash:

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_FAN 8

#define DATA_PIN_FAN_1 6
#define DATA_PIN_FAN_2 7
#define DATA_PIN_FAN_3 8
#define DATA_PIN_FAN_4 9
#define DATA_PIN_FAN_5 10
#define DATA_PIN_FAN_6 5
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[96];
CRGB ledsChannel2[96];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	// 6 fans on channel 1
	//                          FAN_PIN           CRGB array            offset             number of leds per fan
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 3, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 4, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 5, NUMBER_OF_LEDS_PER_FAN);

	// normal strip on channel 2
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 96);
	ledController.addLEDs(0, ledsChannel1, 96);
	ledController.addLEDs(1, ledsChannel2, 96);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

closed 2024-04-20 by Artasdmc