Legion2/CorsairLightingProtocol #372: Is it possible to combine multifan with scale/repeat on a strip?

Artasdmc, 2024-04-20:
This is the simple code adjustment I used, but it doesn't work.

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_FAN 8

#define DATA_PIN_FAN_1 6
#define DATA_PIN_FAN_2 7
#define DATA_PIN_FAN_3 8
#define DATA_PIN_FAN_4 9
#define DATA_PIN_FAN_5 10
#define DATA_PIN_FAN_6 5
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[96];
CRGB ledsChannel2[178];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	// 6 fans on channel 1
	//                          FAN_PIN           CRGB array            offset             number of leds per fan
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 3, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 4, NUMBER_OF_LEDS_PER_FAN);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 5, NUMBER_OF_LEDS_PER_FAN);

	// normal strip on channel 2
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 96);
	ledController.addLEDs(0, ledsChannel1, 96);
	ledController.addLEDs(1, ledsChannel2, 178);
	ledController.onUpdateHook(2, []() {
		CLP::scale(&ledController, 2, 144);
	});
}
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

Legion2, 2024-04-20:
The `ledsChannel2` array must be 178 in size, also `FastLED.addLeds` must use the size of the led strip

Artasdmc, 2024-04-20:
> The `ledsChannel2` array must be 178 in size, also `FastLED.addLeds` must use the size of the led strip

Still the same result, gets halfway to the strip by using 'external rgb light strips', even shorter if I select 6 'rgb led strips'.
 Updated code: 

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUMBER_OF_LEDS_PER_FAN 8

#define DATA_PIN_FAN_1 6
#define DATA_PIN_FAN_2 7
#define DATA_PIN_FAN_3 8
#define DATA_PIN_FAN_4 9
#define DATA_PIN_FAN_5 10
#define DATA_PIN_FAN_6 5
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[48];
CRGB ledsChannel2[178];

CorsairLightingFirmwareStorageEEPROM firmwareStorage;
CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);
FastLEDControllerStorageEEPROM storage;
FastLEDController ledController(&storage);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
// 6 fans on channel 1
// FAN_PIN CRGB array offset number of leds per fan
FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);
FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 3, NUMBER_OF_LEDS_PER_FAN);
FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 4, NUMBER_OF_LEDS_PER_FAN);
FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 5, NUMBER_OF_LEDS_PER_FAN);

// normal strip on channel 2
FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 178);
ledController.addLEDs(0, ledsChannel1, 48);
ledController.addLEDs(1, ledsChannel2, 89);
ledController.onUpdateHook(2, []() {
  CLP::scale(&ledController, 2, 178);
});

}

void loop() {
cHID.update();

if (ledController.updateLEDs()) {
FastLED.show();
}
}

Artasdmc, 2024-04-20:
By changing ledController.addLEDs(1, ledsChannel2, 89); to 178 it does not control anymore.

Legion2, 2024-04-20:
indexes in programming are 0 based, so this is wrong
```c++
ledController.onUpdateHook(2, {
CLP::scale(&ledController, 2, 178);
});
```
there is not channel at index 2, it should be 1

closed 2024-04-23 by Artasdmc