Legion2/CorsairLightingProtocol #377: RP2040 build is broken

EvilGremlin, 2024-09-26:
With arduino ide: 
```
In file included from /home/eg/Arduino/libraries/FastLED/src/FastLED.h:81,
                 from /home/eg/Arduino/libraries/Corsair_Lighting_Protocol/src/FastLEDController.h:18,
                 from /home/eg/Arduino/libraries/Corsair_Lighting_Protocol/src/CorsairLightingProtocol.h:34,
                 from /home/eg/Arduino/libraries/Corsair_Lighting_Protocol/examples/TinyUSB/TinyUSB.ino:16:
/home/eg/Arduino/libraries/FastLED/src/fastspi.h:178:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
  178 | #      pragma message "Forcing software SPI - no hardware SPI for you!"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TinyUSB:31:1: error: 'CorsairLightingProtocolTinyUSBHID' does not name a type; did you mean 'CorsairLightingProtocolController'?
   31 | CorsairLightingProtocolTinyUSBHID cHID(&cLP);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      | CorsairLightingProtocolController
/home/eg/Arduino/libraries/Corsair_Lighting_Protocol/examples/TinyUSB/TinyUSB.ino: In function 'void setup()':
TinyUSB:34:9: error: 'cHID' was not declared in this scope
   34 |         cHID.setup();
      |         ^~~~
/home/eg/Arduino/libraries/Corsair_Lighting_Protocol/examples/TinyUSB/TinyUSB.ino: In function 'void loop()':
TinyUSB:43:9: error: 'cHID' was not declared in this scope
   43 |         cHID.update();
      |         ^~~~
exit status 1
'CorsairLightingProtocolTinyUSBHID' does not name a type; did you mean 'CorsairLightingProtocolController'?
```

With platformio:
```
Executing task: platformio run 

Processing pico (platform: raspberrypi; board: pico; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed @ 4.1.5 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Corsair Lighting Protocol @ 0.15.0
|-- Adafruit TinyUSB Library @ 3.3.4
|-- FastLED @ 3.7.8
Building in release mode
Compiling .pio/build/pico/liba8c/FastLED/FastLED.cpp.o
Compiling .pio/build/pico/liba8c/FastLED/bitswap.cpp.o
In file included from .pio/libdeps/pico/FastLED/src/led_sysdefs.h:38,
                 from .pio/libdeps/pico/FastLED/src/FastLED.h:57,
                 from .pio/libdeps/pico/FastLED/src/FastLED.cpp:2:
.pio/libdeps/pico/FastLED/src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h:4:10: fatal error: hardware/sync.h: No such file or directory
    4 | #include "hardware/sync.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/pico/liba8c/FastLED/colorpalettes.cpp.o
*** [.pio/build/pico/liba8c/FastLED/FastLED.cpp.o] Error 1
In file included from .pio/libdeps/pico/FastLED/src/led_sysdefs.h:38,
                 from .pio/libdeps/pico/FastLED/src/FastLED.h:57,
                 from .pio/libdeps/pico/FastLED/src/colorpalettes.cpp:8:
.pio/libdeps/pico/FastLED/src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h:4:10: fatal error: hardware/sync.h: No such file or directory
    4 | #include "hardware/sync.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/liba8c/FastLED/colorpalettes.cpp.o] Error 1
======================================================================= [FAILED] Took 3.29 seconds =======================================================================

```

closed 2024-09-30 by EvilGremlin

HyperionCoding, 2024-10-31:
So what exactly is this fix? I'm having the same problem but I have no idea what that fix is or where it is.

EvilGremlin, 2024-11-01:
you must use `platform = https://github.com/maxgerhardt/platform-raspberrypi.git`