Legion2/CorsairLightingProtocol #380: Add channels to LNP

TexasTomBF, 2025-02-15:
Great project, got it up and running super fast using a Leonardo beetle mini USB.

This is just a quicks question really, is it possible to just add as many channels as I have Digital ports on my Leonardo, and the iCUE will recognize those? Or is the iCUE limited because of the real product having just two channels.

I am running argb fans and would want to individually control all 9 fans...

Cheers!

Legion2, 2025-02-15:
You can only have two channels, but you can use more pins for your fans, see this example for 6 fans, you can change it for more fans https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples%2FMultipleFans%2FMultipleFans.ino

TexasTomBF, 2025-02-15:
Thanks, luckily my ARGB fans are already daisy chained.

The only thing would be to control them all individually, but that's a limitation in iCUE paired with ARGB devices I guess, no biggie. Thanks again for a great project.

Legion2, 2025-02-15:
In iCUE you can also define the Fans and control them with individual effects. If there is no fan in iCUE which matches the led count of your fans, you can scale the led numbers in Arduino.

closed 2025-03-10 by github-actions[bot]