Legion2/CorsairLightingProtocol #383: Invalid FQBN: 'build.core' and 'build.variant' refer to different platforms: arduino:arduino and SparkFun:promicro

rbf351, 2025-05-13:
When trying to compile and upload to my board I get the following message

Using a Pro Micro and choosing CLP Sparkfun Pro Micro as my board

**Invalid FQBN: 'build.core' and 'build.variant' refer to different platforms: arduino:arduino and SparkFun:promicro

Compilation error: Invalid FQBN: 'build.core' and 'build.variant' refer to different platforms: arduino:arduino and SparkFun:promicro**

rbf351, 2025-05-14:
version 2.3.6

if I change the board type to a CLP Arduino Leonardo there are no errors, and it shows up in iCUE, but windows shows the device as a USB Serial Device

closed 2025-06-05 by github-actions[bot]

Legion2, 2025-07-07:
I tested with Arduino IDE 1.x, 2.x may not work

Legion2, 2025-07-07:
you can get the legacy 1.8.X IDE version on the official website

se-o-matic, 2025-07-09:
the installed versions are:
CLP Board: 0.3.1 (includes the sparkfun micro)
FastLED: 3.10.1
Corsair Lighting Protocol: 0.15.0

Doughnut42, 2026-09-13:
Same.  IDE 2.3.10 states build.core and build.variant are different platforms.  I don't see anything glaringly obvious and even tried deleting the SparksFun:" portion of build.variant in the board definition with no improvement.  Auto detect says an arduino Leonardo is connected so I tried selecting CLP Arduino Leonardo with the expectation that the pinouts would misbehave.  The compiler returned about 20 errors and the board would not upload the code.  

Running the example in IDE 1.8.19 decides it doesn't recognize CorsairLightingProtocol.h as a library.  Clean installs of both have not changed their respective behavior.  I'm new to arduino, so I very well could be doing something wrong, but I am very lost right now.

Any advice would be stelar!

OS: W11

Libraries:
Corsair Lighting Protocol 0.15.0
FastLED 3.10.5

Board manager:
Arduino AVR Boards 1.8.8
Corsair Lighting Protocol Boards 0.3.1
SparkFun AVR Boards 1.1.13

Hardware:
Pro Micro 5V 16 MHz