Legion2/CorsairLightingProtocol #388: CLP with more LEDs per channel (not Icue)

saeedbs, 2025-08-24:
Hey, I've been using 4 CLP boards for about 2 years:  Commander pro - Node pro - LS100 -Node core 
I wanted to mix 2 of them together to have decrease number of my boards so I built new one and increased LEDs per channel to be percise I wanted 276 and 150 for my channels but it didn't work,  I know ICUE doesn't support more LEDs per channel than defaults defined in CLP,  so I'm using OpenRGB and as soon as I set LED numbers in OpenRGB and try to test them,  board crashes  vanishes OR USB device device descriptor fails until I re-plug it 

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [Arduino Pro micro]
 - Device: [tried Lighting Node PRO and LS100]
 - Version of IDE: [Arduino IDE 2.3.6]
 - Version of CLP: [0.3.1]
 - Version of CLP Boards: [0.1.0]
 - Sketch: [tried different skteches]

**Code changes**
Did you changed the code of the library or the sketch?
yes, I changed number of LEDs per channels to 276 and 150 

my board picture :)
[image]

Legion2, 2025-08-24:
Hi, That are too many LEDs, this does not fit into the memory of the board. You need to reduce the memory used, either by decresing the number of the leds in the ledsChannel arrays or in the number of leds configured in iCUE/OpenRGB (ledController.addLEDs). You can keep the numer of physical leds the same but reduce the logical leds and use a scaling function to remap the logical leds to the physical leds. This will save memory will prevent the crashes.

saeedbs, 2025-08-25:
so does lowering it to something like total of 266 LEDs from 350 will fix it or still too many? whats maximum number for micro pro? and about scaling function.... it will repeat effect right? and won't have unique effect for every led?
fyi: chatgpt miss leaded me to believe this board can do ~650 LEDs in total
what other supported board do you suggest with more memory?